DRuggeri/nut_exporter #1: Docker image

c0defre4k, 2020-10-01:
Hi, this looks very promising, great job!

Can you provide a docker image? Readme says something about using this exporter with docker, but there isnt't a Dockerfile. If you could provide an image on hub.docker.com it would be awesome.

DRuggeri, 2020-10-01:
Hey there, @c0defre4k - my bad. I've added the Dockerfile so you can go ahead and build (without needing dependencies on your local box) and then run straight from Docker.

I'll look into adding a push to Docker Hub in the release tooling, but for now, I think this will get you up and running.

c0defre4k, 2020-10-02:
There seems to be a little issue with the `go install` command. Build the image resulted in the following error:
```
Step 8/13 : RUN go install github.com/DRuggeri/nut_exporter
 ---> Running in 3f9691805031
src/github.com/DRuggeri/nut_exporter/collectors/nut_collector.go:5:2: cannot find package "github.com/prometheus/client_golang/prometheus" in any of:
	/usr/local/go/src/github.com/prometheus/client_golang/prometheus (from $GOROOT)
	/app/src/github.com/prometheus/client_golang/prometheus (from $GOPATH)
src/github.com/DRuggeri/nut_exporter/nut_exporter.go:10:2: cannot find package "github.com/prometheus/client_golang/prometheus/promhttp" in any of:
	/usr/local/go/src/github.com/prometheus/client_golang/prometheus/promhttp (from $GOROOT)
	/app/src/github.com/prometheus/client_golang/prometheus/promhttp (from $GOPATH)
src/github.com/DRuggeri/nut_exporter/collectors/nut_collector.go:6:2: cannot find package "github.com/prometheus/common/log" in any of:
	/usr/local/go/src/github.com/prometheus/common/log (from $GOROOT)
	/app/src/github.com/prometheus/common/log (from $GOPATH)
src/github.com/DRuggeri/nut_exporter/nut_exporter.go:12:2: cannot find package "github.com/prometheus/common/version" in any of:
	/usr/local/go/src/github.com/prometheus/common/version (from $GOROOT)
	/app/src/github.com/prometheus/common/version (from $GOPATH)
src/github.com/DRuggeri/nut_exporter/collectors/nut_collector.go:7:2: cannot find package "github.com/robbiet480/go.nut" in any of:
	/usr/local/go/src/github.com/robbiet480/go.nut (from $GOROOT)
	/app/src/github.com/robbiet480/go.nut (from $GOPATH)
src/github.com/DRuggeri/nut_exporter/nut_exporter.go:13:2: cannot find package "gopkg.in/alecthomas/kingpin.v2" in any of:
	/usr/local/go/src/gopkg.in/alecthomas/kingpin.v2 (from $GOROOT)
	/app/src/gopkg.in/alecthomas/kingpin.v2 (from $GOPATH)
The command '/bin/sh -c go install github.com/DRuggeri/nut_exporter' returned a non-zero code: 1
```

Replacing `RUN go install github.com/DRuggeri/nut_exporter` with `RUN cd /app/src/github.com/DRuggeri/nut_exporter && go install` solved the problem.

fnord123, 2020-10-02:
I had the same problem as @c0defre4k.  His fix worked for me. Thanks!

DRuggeri, 2020-10-02:
Argh - sorry about that, folks. I guess I had an older Docker image pulled that it worked OK without changing to the directory. Can confirm that a fresh pull from the hub for golang:buster doesn't work with the Dockerfile.

I've incorporated @c0defre4k's fix into the Dockerfile now. Thanks!

DRuggeri, 2020-10-10:
I've pushed and released the fix. Thanks again!

closed 2020-10-10 by DRuggeri