DRuggeri/nut_exporter #3: Multiple Servers

sshaikh, 2020-12-26:
Thank you for this software, works a treat. I especially like how it doesn't assume to be on the same server as NUT.

To that end it seems useful to be able to leverage multiple servers, so that you only need a single bridge if you have multiple UPS's with local NUT servers.

For this to work you'd probably need a further query parameter, or perhaps use the `instance@server` format that `upsc` uses.

DRuggeri, 2020-12-28:
I really like this idea. I've added it in b92a59b18475ce47899edbbac618cadcda8f5acc as well as released `v1.2.0` incorporating it.

closed 2020-12-28 by DRuggeri

sshaikh, 2020-12-28:
Amazing work, thank you for the quick turnaround!

I'm using the following scrap config to scrape multiple servers in a single job using the new functionality:

```
  - job_name: ups
    static_configs:
      - targets: ['server1','server2'] # nut exporter
    metrics_path: /ups_metrics
    relabel_configs:
      - source_labels: [__address__]
        target_label: __param_server
      - source_labels: [__param_server]
        target_label: instance
      - target_label: __address__
        replacement: nut-exporter.local:9199
```

This will generate two targets of:

http://nut-exporter.local:9199/ups_metrics?server=server1
http://nut-exporter.local:9199/ups_metrics?server=server2

and set the `instance` label of each metric to serverx as appropriate.