DRuggeri/nut_exporter #4: Add all available metrics to default

sshaikh, 2020-12-28:
Instead of having a hard coded list of defaults, include everything available when that parameter is omitted.

I'm not sure if this can be done easily with the go nut library, but 

1) sometimes you don't know what you want tracked until after an event occurs
2) if a new version of NUT is released then we shouldn't need to change our exporter config to pick things up.

DRuggeri, 2020-12-28:
Hi, @sshaikh. This can be accomplished already by just setting the `--nut.vars_enable` to an empty string.

The challenge gets to be cases where NUT is not exposing a metric as a number. [A patch](https://github.com/robbiet480/go.nut/pull/4) was needed for the go.nut library to coax values as closely to go types as possible. This then allowed the nut_exporter to expose anything that appears to be a number or boolean.

Unfortunately, as is the case with `ups.status`, several things returned by NUT are not a number but are strings. Since Prometheus has no mechanism to export strings, those metrics get dropped unless they are handled specially. Obviously, `ups.status` is important enough to handle the string values and a manually managed translation table. Similarly, the `device` metric is coaxed to a set of labels (dropping the uptime value) containing strings because it, too is important.

sshaikh, 2020-12-28:
That's great and makes sense. I was only interested in "output.voltage", so an empty string works for me!

closed 2020-12-28 by sshaikh