DRuggeri/nut_exporter #7: Multiple ups configuration

petshopjke, 2021-05-13:
Hello, I am running two nut servers in docker, each of them monitor UPS connected to host (2 UPS connected via usb to the same host) Can't figure out how to make exporter scrape two NUT servers. Can you help?

DRuggeri, 2021-05-13:
Sure - that should be doable with a small modification to the first example in the README doc as a sample. I think something like this should work fine (note that the `ups` parameter was removed from the example).

```  - job_name: nut-primary
    metrics_path: /ups_metrics
    static_configs:
      - targets: ['exporterserver:9199']
        labels:
          ups:  "primary"
    params:
      server: [ "nutserver1.mydomain.local" ]
  - job_name: nut-secondary
    metrics_path: /ups_metrics
    static_configs:
      - targets: ['exporterserver:9199']
        labels:
          ups:  "secondary"
    params:
      server: [ "nutserver2.mydomain.local" ]
```

You can test if this will work in your network by using a simple curl command:
`curl exporterserver:9199/ups_metrics?server=nutserver1.mydomain.local`

DRuggeri, 2021-06-05:
I'll go ahead and close this out for now, but feel free to ping with follow up questions

closed 2021-06-05 by DRuggeri

theraspb3rry, 2025-11-21:
Hi @DRuggeri, is there a reason why the exporter doesnt loop over all UPS's and add its identifier as a label so you dont need to scrape the exporter twice? would you welcome a patch for such a design? Thanks for this exporter though, I'll be using it regardless :)