DRuggeri/nut_exporter #9: Add UPS Remaining Runtime metric

k0b4lt, 2021-08-20:
I really like your exporter. I was using github.com/cryopie/nut_exporter/ before finding yours. Any chance of adding the remaining runtime as a metric to yours? it's the only thing i'm missing on my dashboard now that I've switched to your exporter

DRuggeri, 2021-08-21:
Hey there, @k0b4lt - thanks for the feedback. Glad to see you're getting some use from the exporter!

This metric actually should already be exported if your UPS supports it. On my TrippLite UPS devices, I see it exported as network_ups_tools_battery_runtime. Have you taken a look at the `--nut.vars_enable` parameter? Tip: You can tell the exporter to just export all numeric variables by setting `--nut.vars_enable=""` on the command line.

I believe I didn't set this one as a metric enabled by default because some UPS devices I've tested with don't support it. Let me know if that works for you - if not, we could take a look at the debug output to see if NUT is exposing it, but maybe the UPS is not providing a numeric result.

k0b4lt, 2021-08-21:
oh hot damn. that's all I needed and now have all kinds of things I can display including my runtimes. sorry I guess I should have reread the ReadMe a few more times and I might have caught that on my own. 

Really awesome exporter. Thanks a lot for developing it.

closed 2021-08-21 by k0b4lt