DRuggeri/nut_exporter #11: container permission issues

jr0dd, 2021-09-06:
When trying to run this container I'm getting permission denied with the ghcr image and dockerhub. Tried 2.2.2 and 2.2.3 and get the same error.

```
MacBook-Pro-2 k8s-gitops % kubectl run -it --rm --restart=Never --image=ghcr.io/druggeri/nut_exporter:2.2.3 nut-exporter
pod "nut-exporter" deleted
pod default/nut-exporter terminated (ContainerCannotRun)
OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "/nut_exporter": permission denied: unknown
```

DRuggeri, 2021-09-06:
Hey there, @jr0dd - thanks for the report! Yesterday was a long day as I learned how to use GitHub Actions across all of my projects and, of course, I made a simple error and forgot to commit/push the working pipeline before rolling the releases 🤦‍♂️ 

I've nuked both releases and rebuilt - everything should be good now!

```
$ docker rmi druggeri/nut_exporter
Untagged: druggeri/nut_exporter:latest
Untagged: druggeri/nut_exporter@sha256:ebea2676ad90788e2ecc53842bca3e4fb84627e3978e367949f7834c11a84bc3
Deleted: sha256:6546176a2d93dc23840e9504f3b4c0fe1142e84585562bd01a66bc5a9b7256f2
Deleted: sha256:0b02176e58d4384b7a2102f9cdfc192f72d8d2c2b8e17ef0fc58d96697fa9c34
$ docker run -it --rm druggeri/nut_exporter
Unable to find image 'druggeri/nut_exporter:latest' locally
latest: Pulling from druggeri/nut_exporter
2260d4fa5bfd: Pull complete
Digest: sha256:ebea2676ad90788e2ecc53842bca3e4fb84627e3978e367949f7834c11a84bc3
Status: Downloaded newer image for druggeri/nut_exporter:latest
INFO[0000] Starting nut_exporter                         source="nut_exporter.go:230"
INFO[0000] Listening on :9199                            source="nut_exporter.go:250"
```

Go ahead and give that a whirl and feel free to reopen the issue if you have problems

closed 2021-09-06 by DRuggeri

jr0dd, 2021-09-06:
It happens. I'll give it a go later today and report back _if_ there is an issue.