DRuggeri/nut_exporter #17: Crash when connecting to Synology NAS

tomswartz07, 2022-02-07:
Hi There!

Trying to connect this exporter to an existing NUT server on a Synology NAS.

I was able to pull out the relevant connection information, and using other tools like `nut-monitor` allow for connections that work.

However, when running the exporter against this server, navigating to the `/ups_metrics`  endpoint triggers a crash with the following log output.

Other possibly relevant info:
- Running on Docker on this affected NAS
- Using the latest available images

```
level=info ts=2022-02-07T01:48:29.297Z caller=nut_exporter.go:167 msg="Authenticating to NUT server"
level=info ts=2022-02-07T01:48:29.297Z caller=nut_exporter.go:233 msg="Starting nut_exporter" version=
level=info ts=2022-02-07T01:48:29.297Z caller=nut_exporter.go:256 msg="Listening on" address=:9199
level=debug ts=2022-02-07T01:50:19.207Z caller=nut_collector.go:45 msg="Connecting to server" server=$IP_ADDRESS
panic: runtime error: index out of range [0] with length 0

goroutine 51 [running]:
github.com/robbiet480/go%2enut.(*Client).GetVersion(...)
	github.com/robbiet480/go.nut@v0.0.0-20211005235800-e810489c32db/nut.go:148
github.com/robbiet480/go%2enut.Connect(0xc000024094, 0x9, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
	github.com/robbiet480/go.nut@v0.0.0-20211005235800-e810489c32db/nut.go:39 +0x38c
github.com/DRuggeri/nut_exporter/collectors.(*NutCollector).Collect(0xc000204640, 0xc0000ae600)
	github.com/DRuggeri/nut_exporter/collectors/nut_collector.go:46 +0x212
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
	github.com/prometheus/client_golang@v1.11.0/prometheus/registry.go:446 +0x1a2
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather
	github.com/prometheus/client_golang@v1.11.0/prometheus/registry.go:538 +0xe8e
level=info ts=2022-02-07T01:50:22.556Z caller=nut_exporter.go:167 msg="Authenticating to NUT server"
level=info ts=2022-02-07T01:50:22.556Z caller=nut_exporter.go:233 msg="Starting nut_exporter" version=
level=info ts=2022-02-07T01:50:22.556Z caller=nut_exporter.go:256 msg="Listening on" address=:9199
```

Any thoughts on what the issue may be?
Happy to run any additional commands or actions to collect information that would be helpful

Thanks!

tomswartz07, 2022-02-07:
Minor followup:

It appears that this crash is caused by per-ip limiting on the Synology side.
It's not super obvious through the Synology DSM interface, but they appear to limit access to the NUT Server/Monitor instances on a per-IP basis.

Adding the explicit container IP address temporarily here allowed the exporter to work.

That being said, it may be the case that there would be a good edge-case for testing and crash prevention here, IMO.

tomswartz07, 2022-02-15:
Thanks for the update! I thought as much as well.

I'll try to get a better picture of what's happening in between/in-front-of the NUT server on Synology, it seems to be a simple IP based block/allow list, but I hope to come up with a reproduction case.

Will follow up as soon as I can.

Thanks again!

DRuggeri, 2022-02-19:
OK, @tomswartz07 - I think we're set. The PR that should avoid this issue has been incorporated upstream, so I just cut a new release with the fixes included. It would be really great if you could disable the IP ACL and see if we can at least catch the error gracefully rather than panic.

tomswartz07, 2022-02-22:
Awesome! 

Just tested it out, and as it happened, the docker container IP address changed as I recreated it, so perfect opportunity.

I can confirm that the latest update handles the issue correctly.
When the IP is not on the allow-list, the following is emitted:

```
level=debug ts=2022-02-22T14:27:58.637Z caller=nut_collector.go:52 msg=Authenticated server=IP_ADDRESS user=USER
level=error ts=2022-02-22T14:27:58.637Z caller=nut_collector.go:58 err="The client’s host and/or authentication details (username, password) are not sufficient to execute the requested command"
```

Issue is resolved, in my opinion!

Thanks again for the amazing work here. I appreciate it.