DRuggeri/nut_exporter #20: Docker connection refused

alteredtech, 2022-04-14:
When attempting to connect to a ups I get that the connection is refused. The docker container starts and says it is authenticating to nut server. Then when I attempt to get the info, the logs show connection refused. I also only have one ups.

This is my docker run command:

`docker run --name nut -p 9199:9199 -d -e NUT_EXPORTER_PASSWORD=somepassword druggeri/nut_exporter --nut.username=admin`

Then these are the logs:

```
ts=2022-04-14T19:03:39.654Z caller=nut_exporter.go:170 level=info msg="Authenticating to NUT server"
ts=2022-04-14T19:03:39.654Z caller=nut_exporter.go:236 level=info msg="Starting nut_exporter" version=2.3.5
ts=2022-04-14T19:03:39.655Z caller=nut_exporter.go:259 level=info msg="Listening on" address=:9199
ts=2022-04-14T19:03:52.168Z caller=nut_collector.go:58 level=error err="dial tcp 127.0.0.1:3493: connect: connection refused"
```

This is how I am testing the endpoint. curl localhost:9199/ups_metrics?ups=nutdev1

DRuggeri, 2022-04-16:
This is very strange - I can't duplicate this by running the exact commands you've pasted
```
hax@hax:~$ docker run --name nut -p 9199:9199 -d -e NUT_EXPORTER_PASSWORD=somepassword druggeri/nut_exporter --nut.username=admin
Unable to find image 'druggeri/nut_exporter:latest' locally
latest: Pulling from druggeri/nut_exporter
e9e7184030a3: Pull complete
Digest: sha256:c3ddd8aa29e775ea83f571f8101fd5c13d1e0e01dffdd7e0d84c38a3cef42337
Status: Downloaded newer image for druggeri/nut_exporter:latest
9287bd9fa33d0016b8e812e90804e584d10235a435358f4075d4d881473c1eae
hax@hax:~$ netstat -an | grep LISTEN
...
tcp        0      0 0.0.0.0:9199            0.0.0.0:*               LISTEN
...
hax@hax:~$ curl 127.0.0.1:9199/metrics
# HELP build_info A metric with a constant '1' value labeled by version, revision, branch, and goversion from which  was built.
# TYPE build_info gauge
build_info{branch="",goversion="go1.15.15",revision="",version=""} 1
...
```

Are you sure the daemon is still running in your container?

DRuggeri, 2022-06-25:
Closing due to lack of response, but feel free to reach out if you can share a reproducible recipe

closed 2022-06-25 by DRuggeri