DRuggeri/nut_exporter #21: Support ups.beeper.status

LXGaming, 2022-06-21:
The blazer driver returns `enabled` / `disabled`, I believe that because it's returning a string this information isn't being exported
https://github.com/networkupstools/nut/blob/3744cb0476820d3bbb6dc3f4ba62feb94673d9b4/drivers/blazer.c#L244-L248

```
caller=nut_collector.go:133 level=debug msg="Export the variable? true"
caller=nut_collector.go:117 level=debug variable_name=ups.beeper.status value=true type=STRING description="UPS beeper status" writeable=false maximum_length=0 original_type=NUMBER
```

DRuggeri, 2022-06-25:
Hi, @LXGaming  - you're correct. Strings are currently ignored, but they don't have to be if they're obviously binary values like you describe. I've added this functionality in be1329db62de0102789b0512ef239f0588b0fb03 and will release soon!

DRuggeri, 2022-06-25:
This has been released as `v2.4.1`

LXGaming, 2022-06-26:
Thanks for adding this functionality, using regex is an awesome way of implementing support for these types of metrics.

I'm currently encountering this issue:
```
ts=2022-06-26T04:15:09.202Z caller=nut_exporter.go:178 level=info msg="Authenticating to NUT server"
ts=2022-06-26T04:15:09.202Z caller=nut_exporter.go:246 level=info msg="Starting nut_exporter" version=2.4.1
ts=2022-06-26T04:15:09.202Z caller=nut_exporter.go:269 level=info msg="Listening on" address=:9199
panic: interface conversion: interface {} is bool, not string

goroutine 44 [running]:
github.com/DRuggeri/nut_exporter/collectors.(*NutCollector).Collect(0xc000210820, 0xc00022e720?)
	github.com/DRuggeri/nut_exporter/collectors/nut_collector.go:197 +0x2d1a
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
	github.com/prometheus/client_golang@v1.11.0/prometheus/registry.go:446 +0xfb
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather
	github.com/prometheus/client_golang@v1.11.0/prometheus/registry.go:538 +0xb0b
```

DRuggeri, 2022-06-27:
Thanks for the heads up, @LXGaming - this seems strange... the underlying library is saying the [variable is a STRING](https://github.com/DRuggeri/nut_exporter/blob/c550e4f7dfaa5b8e766bedd980d00f2e800ca578/collectors/nut_collector.go#L189) but [attempts to use it as a string in a debug statement](https://github.com/DRuggeri/nut_exporter/blob/c550e4f7dfaa5b8e766bedd980d00f2e800ca578/collectors/nut_collector.go#L197) indicates it is actually a boolean. I'll have to look closer at this - maybe ignoring the library type altogether?

DRuggeri, 2022-06-27:
@LXGaming - can you re-run with debug mode enabled? There should be [a few lines leading up to the error](https://github.com/DRuggeri/nut_exporter/blob/c550e4f7dfaa5b8e766bedd980d00f2e800ca578/collectors/nut_collector.go#L141-L149) that would confirm my hunch.

LXGaming, 2022-06-28:
Here is the entire output: ([noticed a minor spelling mistake](https://github.com/DRuggeri/nut_exporter/blob/be1329db62de0102789b0512ef239f0588b0fb03/collectors/nut_collector.go#L132))
```
ts=2022-06-28T06:39:37.337Z caller=nut_exporter.go:178 level=info msg="Authenticating to NUT server"
ts=2022-06-28T06:39:37.338Z caller=nut_exporter.go:246 level=info msg="Starting nut_exporter" version=2.4.1
ts=2022-06-28T06:39:37.338Z caller=nut_exporter.go:269 level=info msg="Listening on" address=:9199
ts=2022-06-28T06:39:44.294Z caller=nut_collector.go:69 level=debug msg="Connecting to server" server=10.0.0.4
ts=2022-06-28T06:39:44.295Z caller=nut_collector.go:72 level=debug msg="Connected to server" server=10.0.0.4
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:88 level=debug msg="Instantiated UPS" name=dynamix
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:127 level=debug msg="UPS info" name=dynamix description="Dynamix Defender 650VA" master=false nmumber_of_logins=1
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:135 level=debug client=0 name=127.0.0.1
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=beeper.toggle description="Toggle the UPS beeper"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=load.off description="Turn off the load immediately"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=load.on description="Turn on the load immediately"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=shutdown.return description="Turn off the load and return when power is back"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=shutdown.stayoff description="Turn off the load and remain off"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=shutdown.stop description="Stop a shutdown in progress"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=test.battery.start description="Start a battery test"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=test.battery.start.deep description="Start a deep battery test"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=test.battery.start.quick description="Start a quick battery test"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:138 level=debug command=test.battery.stop description="Stop the battery test"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:141 level=debug variable_name=battery.charge value=100 type=INTEGER description="Battery charge (percent of full)" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:157 level=debug msg="Export the variable? true"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:141 level=debug variable_name=battery.voltage value=13.6 type=FLOAT_64 description="Battery voltage (V)" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:157 level=debug msg="Export the variable? true"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:141 level=debug variable_name=battery.voltage.high value=13 type=FLOAT_64 description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:157 level=debug msg="Export the variable? true"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:141 level=debug variable_name=battery.voltage.low value=10.4 type=FLOAT_64 description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:157 level=debug msg="Export the variable? true"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:141 level=debug variable_name=battery.voltage.nominal value=12 type=FLOAT_64 description="Nominal battery voltage (V)" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:157 level=debug msg="Export the variable? true"
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:141 level=debug variable_name=device.type value=ups type=STRING description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.304Z caller=nut_collector.go:239 level=debug msg="Export the variable? false" count=17 variables="unsupported value type"
ts=2022-06-28T06:39:44.305Z caller=nut_collector.go:141 level=debug variable_name=driver.name value=blazer_usb type=STRING description="Driver name" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.305Z caller=nut_collector.go:239 level=debug msg="Export the variable? false" count=17 variables="unsupported value type"
ts=2022-06-28T06:39:44.305Z caller=nut_collector.go:141 level=debug variable_name=driver.parameter.langid_fix value=0x0409 type=STRING description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.305Z caller=nut_collector.go:239 level=debug msg="Export the variable? false" count=17 variables="unsupported value type"
ts=2022-06-28T06:39:44.305Z caller=nut_collector.go:141 level=debug variable_name=driver.parameter.pollinterval value=2 type=INTEGER description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2022-06-28T06:39:44.305Z caller=nut_collector.go:239 level=debug msg="Export the variable? false" count=17 variables="unsupported value type"
[... cut ...]
goroutine 30 [running]:
github.com/DRuggeri/nut_exporter/collectors.(*NutCollector).Collect(0xc000095040, 0xc0000b2840?)
	github.com/DRuggeri/nut_exporter/collectors/nut_collector.go:197 +0x2d1a
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
	github.com/prometheus/client_golang@v1.11.0/prometheus/registry.go:446 +0xfb
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather
	github.com/prometheus/client_golang@v1.11.0/prometheus/registry.go:538 +0xb0b
```

DRuggeri, 2022-07-04:
OK, got it. Yeah, it looks like the library says it's a STRING, but it's actually a boolean so it cannot be converted. There's logic elsewhere to handle booleans, so I've just moved the string handling stuff into that same area. Should be more resilient going forward!

Can you give `v2.4.2` a try?

LXGaming, 2022-07-04:
Works perfectly, Thank you

closed 2022-07-04 by LXGaming