DRuggeri/nut_exporter #22: docker compose template

Azertooth, 2022-07-04:
what volume or commands I have to configure in docker compose in order to correctly retrieve the nut package informations?

DRuggeri, 2022-07-09:
Hi, @Azertooth - I'm not very familiar with docker compose, but I do know that Docker isolates the container address space from the host address space by design. It's possible that the container cannot reach out to the host because of this. You can try a few things such as just using the host address space in the exporter container or mapping a NAT rule from the inside to the outside (never tried this one myself).

FWIW, on my home network I run homeassistant and zwave/zigbee integrations in separate containers and the fastest path to getting those containers talking was to just use the host network.

Eroic, 2022-07-21:
HI, 

Try this in your docker compose and remove all "command" section :

    environment:
      - NUT_EXPORTER_SERVER=127.0.0.1 
      - NUT_EXPORTER_USERNAME=myusername
      - NUT_EXPORTER_PASSWORD=mypass
 

It work for me but all other config are by default, 

 I found this relation between command and env. variable for now : 

--nut_server  =   NUT_EXPORTER_SERVER
--nut_username  =   NUT_EXPORTER_USERNAME

NUT_EXPORTER_PASSWORD  (already specified in doc)

We need to find the other : 

--nut_vars_enable
--nut_on_regex
--nut_off_regex
--nut_statuses
--metrics.namespace
--web.listen-address
--web.telemetry-path
--web.exporter-telemetry-path
--web.auth.username
--web.tls.cert_file
--web.tls.key_file
--log.level
--log.format

DRuggeri, 2022-12-17:
@amigthea - the thumbs up in response to @Eroic makes me guess that this worked for you. If that's the case, it'd be really great if you could share a PR that can help future users.

Cheers!

closed 2022-12-17 by DRuggeri

amigthea, 2022-12-21:
> @amigthea - the thumbs up in response to @Eroic makes me guess that this worked for you. If that's the case, it'd be really great if you could share a PR that can help future users.
> 
> Cheers!

hi @DRuggeri, the thumbs up was more of a thank you because in the meantime I had switched to a barebone nut installation, hope his response could be helpful to someone else