DRuggeri/nut_exporter #24: Provide a Load Real Power metric when possible.

dcflachs, 2022-11-19:
For nut devices that provide both a ups.realpower.nominal and ups.load it would be really useful if you could export a synthetic ups.load.realpower metric as ups.realpower.nominal  * (ups.load / 100). This would be really useful for consumers like Netdata which only present metrics they are given and do not provide the ability to create synthetic graphs themselves.

DRuggeri, 2022-12-17:
Thanks for the suggestion @dcflachs  - so, if I understand right, this would calculate the value in the exporter during scrape. That's discouraged based on the usual Prometheus paradigm of how scrape should happen. Plus, it seems that in my configuration, NUT is returning `power.nominal` rather than `realpower.nominal` which may make doing this in the exporter brittle.

... BUT! The Netdata integrations page seems to imply that it will pull metrics from Prometheus (rather than scrape the metric itself). If that is the case, I think a recording rule in Prometheus would work really well.

This one seems to work for me (note, I don't have ups.realpower.nominal, but do have ups.power.nominal):
```
groups:
  - name: realpower
    rules:
    - record: network_ups_tools_realpower
      expr: network_ups_tools_ups_power_nominal * (network_ups_tools_ups_load / 100)
```
The above would go in a rules.yaml file included in your config

In fact, I realized in my setup that I don't even have a current power consumption metric and this suggestion is just what the doctor ordered to start recording it.

Let me know how you get on with this kind of config!

dcflachs, 2022-12-17:
I guess i will have to look for another solution, as one of my goals is to avoid standing up a Prometheus instance. I know it seems crazy as this is a Prometheus exporter, but considering that Netdata can sample Prometheus endpoints directly is seems overkill to add a whole other program just for a single metric.

closed 2023-02-11 by DRuggeri