DRuggeri/nut_exporter #31: Please bump golang.org/x/sys dependency to support riscv64-freebsd

clausecker, 2023-03-11:
Greetings!  This project transitively depends on `golang.org/x/sys`, but a version too old to support riscv64-freebsd is taken.  Please bump the dependency using e.g.

    go get golang.org/x/sys@latest

to fix the build on this new architecture.

DRuggeri, 2023-03-25:
Hi, @clausecker - no problem. I've updated go.mod to 1.18 minimum and included this dependency. On a whim, I attempted to compile with `GOOS=freebsd` and `GOARCH=riskv64` but my runtime told me that's not a valid GOARCH. I'm curious which version you're using?

closed 2023-03-25 by DRuggeri

clausecker, 2023-03-25:
Hi @DRuggeri, you need go 1.20 for this target to be available.  The minimum go.mod version required for the new `golang.org/x/sys` dependency to build is 1.17.