DRuggeri/nut_exporter #32: No verification of client TLS certs?

otakup0pe, 2023-03-16:
Stumbled across this today while rolling some expired certs. The nut exporter doesn't seem to do any kind of verification at least as it pertains to the client cert expiration date. I think that #13 would address this issue.

DRuggeri, 2023-03-25:
Hi, @otakup0pe - thanks! You're correct, the exporter today does not support client certificate as an authentication mechanism. The toolkit referenced in #13 does seem to support this as a feature, but would cause a breaking change on command line params. May be worth it - I need to look again at the toolkit to weight the benefits.

DRuggeri, 2023-03-25:
The toolkit looks like it can add this functionality and I'd say is worth the breaking changes. I requested one quick fix to the toolkit so docs are clear: https://github.com/prometheus/exporter-toolkit/pull/145

Once merged (or help message sufficiently improved), I'll pull it into this exporter

DRuggeri, 2023-03-26:
It has now been incorporated and released as 3.0 - have a look at the release notes for the link to the config file params you can use to enable MTLS auth

closed 2023-03-26 by DRuggeri