DRuggeri/nut_exporter #36: Struggling with installation, don't see the data in the web portal

Jsalas424, 2023-05-28:
I'm trying to install v3.0.0.0 on Debian Bullseye and can't seem to get it work. 

Here's what I attempted:

I downloaded the [binaries here](https://github.com/DRuggeri/nut_exporter/releases/download/v3.0.0/nut_exporter-v3.0.0-linux-amd64) 

```
root@NodeB:~# curl -O https://github.com/DRuggeri/nut_exporter/releases/download/v3.0.0/nut_exporter-v3.0.0-linux-amd64
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0

root@NodeB:~# mv nut_exporter-v3.0.0-linux-amd64 nut_exporter

root@NodeB:~# ./nut_exporter
-bash: ./nut_exporter: Permission denied

root@NodeB:~# chmod +x nut_exporter
root@NodeB:~# ./nut_exporter
```

I then expected to see data at http://192.168.1.25:9199/ups_metrics?ups=TrippLiteUPS but I don't. My nut-client is running and I'm getting data:

```
root@NodeB:~# upsc TrippLiteUPS@localhost
Init SSL without certificate database
battery.voltage: 13.70
battery.voltage.nominal: 12
device.mfr: Tripp Lite
device.model: TRIPP LITE SMART500RT1U
device.type: ups
driver.name: tripplite_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.29
input.voltage: 116
input.voltage.maximum: 124
input.voltage.minimum:  87
input.voltage.nominal: 120
outlet.1.desc: Load 1
outlet.1.id: 1
outlet.1.switch: 1
outlet.1.switchable: 1
outlet.2.desc: Load 2
outlet.2.id: 2
outlet.2.switchable: 0
ups.debug.D: 00 76 00 89 0d 00 00 '.v.....'
ups.debug.L: 0c 00 06 07 58 58 0d '....XX.'
ups.debug.load_banks: 1
ups.debug.M: 00 57 00 7c 0d 00 00 '.W.....'
ups.debug.P: 30 30 35 30 30 58 0d '00500X.'
ups.debug.S: 01 04 00 00 64 00 0d '....d..'
ups.debug.T: 21 00 02 58 01 58 0d '...X.X.'
ups.debug.V: 02 00 0c 01 58 58 0d '....XX.'
ups.delay.shutdown: 64
ups.firmware: F29AF.B
ups.firmware.aux: protocol 3005
ups.id: 65535
ups.mfr: Tripp Lite
ups.model: TRIPP LITE SMART500RT1U
ups.power.nominal: 500
ups.productid: 0001
ups.status: OL
ups.temperature: 0
ups.vendorid: 09ae

root@NodeB:~# systemctl status nut-client
● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-05-28 18:38:24 EDT; 24min ago
   Main PID: 1252374 (upsmon)
      Tasks: 2 (limit: 38378)
     Memory: 964.0K
        CPU: 44ms
     CGroup: /system.slice/nut-monitor.service
             ├─1252373 /lib/nut/upsmon
             └─1252374 /lib/nut/upsmon

May 28 18:39:59 TracheNodeB upsmon[1252374]: Communications with UPS TrippLiteUPS@localhost lost
May 28 18:40:04 TracheNodeB upsmon[1252374]: Communications with UPS TrippLiteUPS@localhost established
May 28 18:44:49 TracheNodeB upsmon[1252374]: Poll UPS [TrippLiteUPS@localhost] failed - Data stale
May 28 18:44:49 TracheNodeB upsmon[1252374]: Communications with UPS TrippLiteUPS@localhost lost
May 28 18:44:54 TracheNodeB upsmon[1252374]: Communications with UPS TrippLiteUPS@localhost established
May 28 18:46:00 TracheNodeB systemd[1]: nut-monitor.service: Supervising process 1252374 which is not ou>
May 28 19:02:04 TracheNodeB upsmon[1252374]: Poll UPS [TrippLiteUPS@localhost] failed - Data stale
May 28 19:02:04 TracheNodeB upsmon[1252374]: Communications with UPS TrippLiteUPS@localhost lost
May 28 19:02:09 TracheNodeB upsmon[1252374]: Poll UPS [TrippLiteUPS@localhost] failed - Data stale
May 28 19:02:14 TracheNodeB upsmon[1252374]: Communications with UPS TrippLiteUPS@localhost established
```

pichouk, 2024-10-20:
Did you find the solution @Jsalas424 ? Same setup here and same issue

DRuggeri, 2024-11-07:
Thanks - can you share the output of the nut_exporter run? Debug mode would be the most helpful. The settings otherwise seem fine since upsc is able to hit the localhost endpoint and list things.

pichouk, 2024-11-12:
I'm so sorry I saw your message, then read again the README in order to enable debug mode to give you more insights and then I noticed this : metrics are exposed on `/ups_metrics`.

100% my fault, it works great sorry for the noise and thanks for your exporter :smile:

closed 2025-01-10 by DRuggeri