DRuggeri/nut_exporter #37: Failure instantiating the UPS - err="error reading response: EOF"

Jsalas424, 2023-07-30:
Similar to issue #27, I am getting an EOF error when running nut_exporter. Unlike the other error, my data does seem to be coming in okay. 

[image]

Here's the scraper config:
```
  - job_name: 'Power'
    metrics_path: /ups_metrics
    static_configs:
      - targets: ['192.168.1.24:9199'] #CyberPower Free Standing UPS
      - targets: ['192.168.1.25:9199'] #TrippLite Rack UPS
```

Here is the debug output:

There nothing about `duplicate metrics collector registration attempted` nor invalid UPS names.

Since data seems to be coming in, I guess I'm not overly concerned but thought i'd report it nontheless. 

```
root@TracheNodeB:~# ./nut_exporter-v2.5.1-linux-arm --log.level=debug
-bash: ./nut_exporter-v2.5.1-linux-arm: No such file or directory
root@TracheNodeB:~# nut_exporter --log.level=debug
ts=2023-07-30T01:13:46.801Z caller=nut_exporter.go:214 level=info msg="Starting nut_exporter" version=3.0.0
ts=2023-07-30T01:13:46.801Z caller=tls_config.go:238 level=info msg="Listening on" address=[::]:9199
ts=2023-07-30T01:13:46.802Z caller=tls_config.go:241 level=info msg="TLS is disabled." http2=false address=[::]:9199
ts=2023-07-30T01:13:48.605Z caller=nut_exporter.go:115 level=info msg="Creating new registry, handler, and collector for UPS `127.0.0.1/`"
ts=2023-07-30T01:13:48.605Z caller=nut_collector.go:84 level=debug msg="Connecting to server" server=127.0.0.1
ts=2023-07-30T01:13:48.605Z caller=nut_collector.go:96 level=debug msg="Connected to server" server=127.0.0.1
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:125 level=debug msg="Obtained list of UPS devices"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:128 level=debug msg="UPS name detection" name=TrippLiteUPS
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:161 level=debug msg="UPS info" name=TrippLiteUPS description="Rack Mounted Tripp Lite" master=false nmumber_of_logins=1
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:169 level=debug client=0 name=127.0.0.1
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:172 level=debug command=load.off description="Turn off the load immediately"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:172 level=debug command=load.on description="Turn on the load immediately"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:172 level=debug command=reset.input.minmax description="Reset minimum and maximum input voltage status"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:172 level=debug command=shutdown.return description="Turn off the load and return when power is back"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:172 level=debug command=test.battery.start description="Start a battery test"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=battery.voltage value=13.7 type=FLOAT_64 description="Battery voltage (V)" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:191 level=debug msg="Export the variable? true"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=battery.voltage.nominal value=12 type=INTEGER description="Nominal battery voltage (V)" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:191 level=debug msg="Export the variable? true"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=device.mfr value="Tripp Lite" type=STRING description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=device.model value="TRIPP LITE SMART500RT1U" type=STRING description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=device.type value=ups type=STRING description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=driver.name value=tripplite_usb type=STRING description="Driver name" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=driver.parameter.pollinterval value=15 type=INTEGER description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=driver.parameter.port value=auto type=STRING description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=driver.parameter.synchronous value=no type=STRING description="Description unavailable" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=driver.version value=2.7.4 type=NUMBER description="Driver version - NUT release" writeable=false maximum_length=0 original_type=
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=driver.version.internal value=0.29 type=FLOAT_64 description="Internal driver version" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=input.voltage value=115 type=INTEGER description="Input voltage (V)" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:191 level=debug msg="Export the variable? true"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=input.voltage.maximum value=124 type=INTEGER description="Maximum incoming voltage seen (V)" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=input.voltage.minimum value=87 type=INTEGER description="Minimum incoming voltage seen (V)" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=input.voltage.nominal value=120 type=INTEGER description="Nominal input voltage (V)" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:191 level=debug msg="Export the variable? true"
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:175 level=debug variable_name=outlet.1.desc value="Load 1" type=STRING description="Outlet description" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.784Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
[... 26 lines cut ...]
[... cut ...]
ts=2023-07-30T01:13:48.785Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
ts=2023-07-30T01:13:48.785Z caller=nut_collector.go:175 level=debug variable_name=ups.vendorid value=09ae type=STRING description="Vendor ID for USB devices" writeable=false maximum_length=0 original_type=NUMBER
ts=2023-07-30T01:13:48.785Z caller=nut_collector.go:268 level=debug msg="Export the variable? false" count=7 variables="unsupported value type"
```

Jsalas424, 2023-07-30:
I think in my situation, this is due to a "Stale" issue. My logs follow this pattern:

```
Jul 30 05:24:31 Server upsmon[2058740]: Poll UPS [TrippLiteUPS@localhost] failed - Driver not connected
Jul 30 05:24:35 Server nut_exporter[2064247]: ts=2023-07-30T09:24:35.650Z caller=nut_collector.go:115 level=error msg="Failure instantiating the UPS" name=TrippLiteUPS err="error reading response: EOF"
```

I had `/etc/nut/ups.conf` configured with pollinterval = 1 but I guess that didn't help with the stale data issue. 

```
maxretry = 3

[TrippLiteUPS]
        driver = tripplite_usb
        port = auto
        desc = "Rack Mounted Tripp Lite"
        pollinterval = 1
```

I've since updated `/etc/nut/upsmon.conf` to include `DEADTIME 25` and `MAXAGE 25`. If `Failure instantiating the UPS` pops up again when data isn't stale, there's something else going on.

Jsalas424, 2023-07-30:
I think this is actually a connectivity issue between the UPS and NUT, I've collected some data to this effect and opened a ticket: https://github.com/networkupstools/nut/issues/2001

DRuggeri, 2023-11-29:
Thanks, @Jsalas424 - checking back in, were you able to get to the root of it? I do agree that an unexpected EOF would indicate an issue communicating with the UPS. On a connectivity issue use case, this would be stemming from the NUT client library expecting a greeting from the server

Jsalas424, 2023-12-02:
> Thanks, @Jsalas424 - checking back in, were you able to get to the root of it? I do agree that an unexpected EOF would indicate an issue communicating with the UPS. On a connectivity issue use case, this would be stemming from the NUT client library expecting a greeting from the server

I'm pretty sure this was a nut-client thing. Nut 2.8.0 resolved a lot of connectivity issues. I documented this across 2 tickets with different UPS's https://github.com/networkupstools/nut/issues/2121 and https://github.com/networkupstools/nut/issues/2001

closed 2023-12-02 by Jsalas424