DRuggeri/nut_exporter #38: Voltages and runtime not detected

kimtore, 2023-10-18:
Hi, first of all thanks for this exporter.

It works well except that only battery charge, load, model info, and device status is scraped. Everything else is ignored. In particular, I would like to get the runtime, input voltage, and output voltage.

I'm using version 3.0.0.

Here is the data scraped from the exporter:
```
$ curl -s localhost:9199/ups_metrics | grep '^network'
network_ups_tools_battery_charge 100
network_ups_tools_device_info{contact="",description="",location="",macaddr="",mfr="EATON",model="Eaton 3S 550",part="",serial="0",type="ups"} 1
network_ups_tools_ups_load 15
network_ups_tools_ups_status{flag="ALARM"} 1
network_ups_tools_ups_status{flag="BOOST"} 0
network_ups_tools_ups_status{flag="BYPASS"} 0
network_ups_tools_ups_status{flag="CAL"} 0
network_ups_tools_ups_status{flag="CHRG"} 0
network_ups_tools_ups_status{flag="DISCHRG"} 0
network_ups_tools_ups_status{flag="FSD"} 0
network_ups_tools_ups_status{flag="HB"} 0
network_ups_tools_ups_status{flag="LB"} 0
network_ups_tools_ups_status{flag="OB"} 0
network_ups_tools_ups_status{flag="OFF"} 0
network_ups_tools_ups_status{flag="OL"} 1
network_ups_tools_ups_status{flag="OVER"} 0
network_ups_tools_ups_status{flag="RB"} 0
network_ups_tools_ups_status{flag="SD"} 0
network_ups_tools_ups_status{flag="TRIM"} 0
```
However, UPSC gives more detail:
```
$ upsc eaton
battery.charge: 100
battery.charge.low: 20  
battery.runtime: 1406             
battery.type: PbAc   
device.mfr: EATON   
device.model: Eaton 3S 550
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.parameter.vendorid: 0463
driver.version: 2.7.4
driver.version.data: MGE HID 1.40
driver.version.internal: 0.41
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: yes                                                                                                                                                                                    outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 3
outlet.2.status: off
outlet.2.switchable: yes
outlet.desc: Main Outlet
outlet.id: 1
outlet.switchable: no
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.alarm: Battery voltage too low!
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02
ups.load: 15
ups.mfr: EATON
ups.model: Eaton 3S 550
ups.power.nominal: 550
ups.productid: ffff
ups.serial: 000000000
ups.status: ALARM OL
ups.timer.shutdown: 0
ups.timer.start: 0
ups.vendorid: 0463
```

grndhog3rd, 2023-11-09:
Either pass (side note it does not appear input.voltage is available to you based on the list you provided.

`--nut.vars_enable=`
Setting the nut.vars_enable parameter to an empty string will cause all numeric variables to be exported
or
`--nut.vars_enable=battery.charge,battery.runtime,battery.voltage,battery.voltage.nominal,input.voltage,input.voltage.nominal,output.voltage,ups.load,ups.status`

The default ones are
`battery.charge,battery.voltage,battery.voltage.nominal,input.voltage,input.voltage.nominal,ups.load,ups.status`

kimtore, 2023-11-09:
Thank you for your help.
Setting `nut.vars_enable` to an empty string didn't help at all. The parameter had to be set explicitly, and then it worked.

closed 2023-11-09 by kimtore

UntouchedWagons, 2023-12-09:
I tried adding `battery.runtime` but battery runtime is still not available via `network_ups_tools_battery_runtime`. My ScrapeConfig:

```
apiVersion: monitoring.coreos.com/v1alpha1
kind: ScrapeConfig
metadata:
  name: exporter-nut
  namespace: monitoring
spec:
  staticConfigs:
    - targets:
        - exporter-nut.monitoring.svc.cluster.local:9199
      labels:
        ups: "OR1500LCDRT2U"
  metricsPath: /ups_metrics
  params:
    server: 
      - "nut.internal.untouchedwagons.com"
    ups: 
      - "OR1500LCDRT2U"
    vars_enable:
      - "battery.charge,battery.runtime,battery.voltage,battery.voltage.nominal,input.voltage,input.voltage.nominal,output.voltage,ups.load,ups.status"
```

Battery runtime is listed in the output of `upsc` even when the UPS has wall power.