DRuggeri/nut_exporter #42: Question on configuration file

tgarneau, 2023-11-27:
you mention the use of a configuration file here: [https://github.com/DRuggeri/nut_exporter/blob/master/README.md#example-scrape-configurations](url)

Where should that config file be placed/mounted when using the Docker container and does it need a particular name?

DRuggeri, 2023-11-30:
Hi, @tgarneau - that section of the readme is intended to describe how you configure Prometheus to scrape this exporter. I've made a small tweak to the section to call it "Example Prometheus Scrape Configurations" so it's hopefully a bit more clear. Does that help?

tgarneau, 2023-11-30:
That does help. Thank you for that. A big part of the problem is that I'm inexperienced with Prometheus and that I came in with the poor assumption that the nut server address and ups parameters would be directly applied to this container. Looking more closely at the example configs I see the -job_name parameter which I now know to associate with the Prometheus scrape config. Thanks again for tweaking the documentation.

closed 2023-11-30 by tgarneau