DRuggeri/nut_exporter #43: Server and Port environment variables conflicting

LXGaming, 2023-11-29:
My container is no longer starting due to this error `nut_exporter: error: strconv.ParseFloat: parsing "192.168.1.2": invalid syntax, try --help`, I believe it's because the recently added port option is attempting to parse the `NUT_EXPORTER_SERVER` environment variable as a port when it's already being used for hostname / IP address.

https://github.com/DRuggeri/nut_exporter/blob/15f476a8c004304cbb687dd01fc8c250cb364997/nut_exporter.go#L26-L32

DRuggeri, 2023-11-30:
You're right! I just pushed a fix in acbcd5c7d575d365f9cdce485394b57321c00fcf and am pushing a release now. Thanks for the report!

LXGaming, 2023-11-30:
Thank you for the quick fix, you might want to update the README as well.

https://github.com/DRuggeri/nut_exporter/blob/15f476a8c004304cbb687dd01fc8c250cb364997/README.md?plain=1#L202

DRuggeri, 2023-11-30:
Spot on! This is also done.

DRuggeri, 2023-11-30:
Hi, @gyulaasztalos - I can't duplicate that. Can you share more details in a new issue? I just ran the exporter like so, and was able to scrape metrics:
```
./nut_exporter --nut.server upsmon.local
ts=2023-11-30T12:49:42.508Z caller=nut_exporter.go:226 level=info msg="Starting nut_exporter" version=testing
ts=2023-11-30T12:49:42.508Z caller=tls_config.go:238 level=info msg="Listening on" address=[::]:9199
ts=2023-11-30T12:49:42.508Z caller=tls_config.go:241 level=info msg="TLS is disabled." http2=false address=[::]:9199
^[+ts=2023-11-30T12:50:08.174Z caller=nut_exporter.go:126 level=info msg="Creating new registry, handler, and collector for UPS `upsmon.local:3493/`"
```

It functions the same if I pass the server target in ENV as well - in all cases, the default of 3493 seems to be functioning.

gyulaasztalos, 2023-11-30:
> Hi, @gyulaasztalos - I can't duplicate that. Can you share more details in a new issue? I just ran the exporter like so, and was able to scrape metrics:
>
> ```
> ./nut_exporter --nut.server upsmon.local [...]

Sorry, my bad. I'm using the docker image with the latest tag. and just realised there was a 3.1.0 release yesterday. I had the problem with that one (the container couldn't start):
nut_exporter: error: strconv.ParseFloat: parsing "raspberrypi.local.example.com": invalid syntax, try --help

So it looks like when I added the new environment variable to my docker-compose.yml it pulled the newer 3.1.1 image and that's what really fixed my issue, not the new env var.

Thank you for your help and all the hard work.