DRuggeri/nut_exporter #46: Port added unconditionally

ColCh, 2024-05-18:
Hello! Thank you very much for this exporter

I'm trying to launch it, using prebuilt binary

```
nut_exporter --version
3.1.1
```

looks like it's trying to add port twice:

```
nut_exporter \
                                        --web.listen-address="myupsserver.local:9199" \
                                        --log.level=debug \
                                        --nut.username="admin"
ts=2024-05-18T10:58:48.752Z caller=nut_exporter.go:156 level=info msg="Authenticating to NUT server"
ts=2024-05-18T10:58:48.753Z caller=nut_exporter.go:226 level=info msg="Starting nut_exporter" version=3.1.1
ts=2024-05-18T10:58:48.753Z caller=tls_config.go:238 level=info msg="Listening on" address=myupsserver.local:9199
ts=2024-05-18T10:58:48.753Z caller=tls_config.go:241 level=info msg="TLS is disabled." http2=false address=myupsserver.local:9199
ts=2024-05-18T10:58:49.518Z caller=nut_exporter.go:126 level=info msg="Creating new registry, handler, and collector for UPS `myupsserver.local:9199:3493/`"
ts=2024-05-18T10:58:49.518Z caller=nut_collector.go:85 level=debug msg="Connecting to server" server=myupsserver.local:9199 port=3493
ts=2024-05-18T10:58:49.518Z caller=nut_collector.go:88 level=error err="address myupsserver.local:9199:3493: too many colons in address"
^C⏎
```

I tried anything: specifying server / serverport manually, removing all args etc. Looks like port is always added

this one looks similiar to #39

DRuggeri, 2024-06-13:
Hey there, @ColCh. Sorry for the delay getting back to you, but you should be able to use the `--nut.serverport` argument for this. The output looks a little confusing - I see that you are setting web.listen-address, but not attempting to set the NUT server. This made me detect a bug in the handling of environment variables, but the command line argument should work OK. Have you tried that one?

ColCh, 2024-06-15:
Hello! :) oh, no troubles here, thank you for answering

upsd is running at default port,

```
127.0.0.1:3493       0.0.0.0:*    users:(("upsd",pid=699513,fd=4))
```

and nut_exporter uses this as default value for nut server (`nut_exporter --help`):

```
      --nut.server="127.0.0.1"   Hostname or IP address of the server to connect to. ($NUT_EXPORTER_SERVER)
                                 ($NUT_EXPORTER_SERVER)
      --nut.serverport=3493      Port on the NUT server to connect to. ($NUT_EXPORTER_SERVERPORT)
                                 ($NUT_EXPORTER_SERVERPORT)
```

as for specifying it explicitly, I tried running it like this (please note, I use env var `$NUT_EXPORTER_PASSWORD` for password)

```
root@ASUS-MINIPC ~ [SIGINT]# nut_exporter \
                                     --web.listen-address="10.20.0.6:9199" \
                                     --log.level=debug \
                                     --nut.server="127.0.0.1" \
                                     --nut.serverport="3493" \
                                     --nut.username="admin"
ts=2024-06-15T15:55:38.149Z caller=nut_exporter.go:156 level=info msg="Authenticating to NUT server"
ts=2024-06-15T15:55:38.149Z caller=nut_exporter.go:226 level=info msg="Starting nut_exporter" version=3.1.1
ts=2024-06-15T15:55:38.149Z caller=tls_config.go:238 level=info msg="Listening on" address=10.20.0.6:9199
ts=2024-06-15T15:55:38.149Z caller=tls_config.go:241 level=info msg="TLS is disabled." http2=false address=10.20.0.6:9199
ts=2024-06-15T15:55:44.338Z caller=nut_exporter.go:126 level=info msg="Creating new registry, handler, and collector for UPS `127.0.0.1:3493/myups`"
ts=2024-06-15T15:55:44.338Z caller=nut_collector.go:302 level=debug msg="Connecting to server and verifying `myups` is a valid UPS name" server=127.0.0.1
ts=2024-06-15T15:55:44.516Z caller=nut_collector.go:326 level=debug msg="UPS name detection" name=myups
ts=2024-06-15T15:55:44.516Z caller=nut_collector.go:332 level=debug msg="Validity result for UPS named `myups`" valid=true
```

-------

oh my gosh , I don't know what happened... may be I messed something and defined some env var previously.. but now this works as expected

many thanks! I will close this issue immediately. Strange things could happen :) let's say it was "incorrect phase of the moon" :D 

have a good day!

closed 2024-06-15 by ColCh