DRuggeri/nut_exporter #47: [Feat.] Config file

sleepy-nols, 2024-06-15:
Hi, I was playing around with this awesome exporter and didn't find any way to read the configuration from a config file, just the command line arguments. Am I blind or is this not supported? I have found `--web.config.file="" ` but I suppose this is just for the web config.

DRuggeri, 2024-09-25:
Hi there, @sleepy-nols. You're correct, it accepts either environment variables or command line parameters only. Basically, it uses the Kingpin library which is a CLI parser.

sleepy-nols, 2024-12-23:
okay, found the environment variables working for me. 

> it accepts either environment variables or command line parameters

does this mean cli arguments are discarded when environmental variables are used?

DRuggeri, 2025-01-10:
Close - just the other way around. The library used by the exporter will use the environment entry if present, but if a command line flag is also provided, the command line flag is preferred. This way you can run several exporters on the same machine (as an example) with the same environment, but individually change each daemon. In that case, all exporters export the same variables due to a common NUT_EXPORTER_VARIABLES environment flag, but each exporter has a different --web.listen-address to avoid port conflicts.

closed 2025-01-10 by DRuggeri