DRuggeri/nut_exporter #50: Can get ups runtime because "unsupported value type"

zoic21, 2024-11-01:
Hello,

I got an issue to get battery runtime, I have an error with "unsupported value type" :

```
ts=2024-11-01T14:05:46.715Z caller=nut_collector.go:176 level=debug variable_name=battery.runtime value=1952 type=INTEGER description="Battery runtime (seconds)" writeable=false maximum_length=0 original_type=NUMBER
ts=2024-11-01T14:05:46.715Z caller=nut_collector.go:269 level=debug msg="Export the variable? false" count=8 variables="unsupported value type"
```

Here my configuration : 

```
services:
  nut-exporter:
    image: druggeri/nut_exporter
    container_name: nut-exporter
    restart: always
    ports:
      - 9199:9199
    command:
      - --nut.vars_enable="battery.runtime,battery.charge,battery.voltage,battery.voltage.nominal,input.voltage,input.voltage.nominal,ups.load,ups.status"
      - --nut.serverport=3493
      - --log.level=debug
```

I don't understand why it's did that, I think I misconfiguring something but I can't find what....

Thanks in advance

mmeija, 2024-11-17:
I think i'm seeing the same behaviour
nut_exporter-v3.1.1-linux-amd64
I think it works as expected with native binary? 
docker does strange things with the nut.vars_enable  - I don't know
in either case without the nut.vars_enable  I don't get battery.runtime - with the vars set in the docker i get battery.runtime but lose ups.status?

here's docker with no nut.vars_enable
```
docker compose:
services:
  nut-exporter:
    image: druggeri/nut_exporter
    container_name: nut-exporter
    restart: always
    ports:
      - 9199:9199
    command:
      - --log.level=debug
      - --nut.server=10.1.4.20
```
[debug_no_vars.txt](https://github.com/user-attachments/files/17792502/debug_no_vars.txt)

and with:
```
services:
  nut-exporter:
    image: druggeri/nut_exporter
    container_name: nut-exporter
    restart: always
    ports:
      - 9199:9199
    command:
      - --nut.vars_enable="battery.runtime,battery.charge,battery.charge.low,battery.charge.warning,battery.runtime,battery.temperature,battery.voltage,battery.voltage.nominal,input.transfer.high,input.transfer.low,ups.beeper.status,input.voltage,input.voltage.nominal,ups.load,ups.status"
      - --log.level=debug
      - --nut.server=10.1.4.20
```
[debug_vars.txt](https://github.com/user-attachments/files/17792513/debug_vars.txt)

`./nut_exporter --nut.server="10.1.4.20" --log.level=debug`
[native_no_vars.txt](https://github.com/user-attachments/files/17792524/native_no_vars.txt)

`./nut_exporter --nut.server="10.1.4.20" --log.level=debug --nut.vars_enable="battery.runtime,battery.charge,battery.charge.low,battery.charge.warning,battery.runtime,battery.temperature,battery.voltage,battery.voltage.nominal,input.transfer.high,input.transfer.low,ups.beeper.status,input.voltage,input.voltage.nominal,ups.load,ups.status"`
[native_vars.txt](https://github.com/user-attachments/files/17792525/native_vars.txt)

kennethso168, 2024-12-22:
Hey all, I just started using the docker container and maybe I can share my solution on this.

I first tried a docker-comopse file without the `--nut.vars_enable` flag and with `--nut.server="10.20.30.40"`. The logs would say something like `level=error err="lookup \"10.20.30.40\": no such host"`. I removed the double quotes in the docker-compose file as `--nut.server=10.20.30.40` and it worked then.

Afterwards I customized the `--nut.vars_enable` flag with the `battery.runtime` variable added. The flag was put without double quotes and I can see that it also worked with the `battery.runtime` successfully exported. I guess docker will quote any double quotes in the command block literally, causing `nut_exporter` to fail to parse the provided `--nut.vars_enable` and fall back to the default. So maybe try putting those values without double quotes and see if this solves your issue.

My working docker-compose file for reference (also with the `webnut` container on the same stack)

```
version: "3.5"
services:
  webnut:
    image: edgd1er/webnut:latest@sha256:cb2c70ec1d0b584c41f4a73ca92c0920c710699f25c17538d012d20f39645fd1
    restart: unless-stopped
    env_file:
      - .env
    environment:
      UPS_HOST: "10.20.30.40"
      UPS_PORT: "3493"
      UPS_USER: "monuser"
      # UPS_PASSWORD in .env
    ports:
     - "6543:6543"
  nut-exporter:
    image: druggeri/nut_exporter:3.1.1
    restart: unless-stopped
    env_file:
      - .env
    ports:
      - "9199:9199"
    # NUT_EXPORTER_PASSWORD in .env
    command: 
      - --nut.server=10.20.30.40
      - --nut.username=monuser
      - --nut.vars_enable=battery.charge,battery.voltage,battery.voltage.nominal,battery.runtime,input.voltage,input.voltage.nominal,ups.load,ups.status

```

DRuggeri, 2025-01-10:
I've been digging around a bit and can't seem to figure out the source of how the exporter got the string "unsupported value type". At first I thought it was coming from Go's [JSON marshaling](https://go.dev/src/encoding/json/encode.go) (which I assumed Docker is doing internally between components) but that didn't pan out. I checked also if the [YAML](https://github.com/go-yaml/yaml/tree/v3.0.1) marshalling/unmarshalling may have been the culprit, but came up empty as well.

I'd like to add a warning for the user if we see something like this, but understanding the source is important in doing so. If the string "unsupported value type" may show up as a value in *any* command line param, it'd be a bit choresome to add a safety net.

Any thoughts?