DRuggeri/nut_exporter #52: network_ups_tools_ups_status variable is missing when using NUT_EXPORTER_VARIABLES

varac, 2024-12-15:
I am using this `docker-compose.yaml` file: 

```yaml
services:
  nut-exporter:
    # https://github.com/DRuggeri/nut_exporter/pkgs/container/nut_exporter
    image: ghcr.io/druggeri/nut_exporter:3.1.1
    hostname: nut-exporter
    container_name: nut-exporter
    restart: unless-stopped
    ports:
      - 10.27.65.51:9199:9199/TCP
    environment:
      - NUT_EXPORTER_SERVER=10.27.65.51
      - NUT_EXPORTER_USERNAME=monitor
      - NUT_EXPORTER_PASSWORD=...
```

Everything work fine, the metrics `network_ups_tools_ups_status` is returned for different lables. 

But I'd like to get the content of more variables, so I set: 

    NUT_EXPORTER_VARIABLES="ups.realpower,ups.load,output.voltage,outlet.power,battery.runtime,battery.charge,battery.charge.low,battery.runtime.low"

With this, the metric  `network_ups_tools_ups_status` is missing completely.
How can I get more variables *and* have still the `network_ups_tools_ups_status` metric ?

maybesheforgot, 2024-12-20:
In my compose file, instead of `NUT_EXPORTER_VARIABLES` I'm using `NUT_VARIABLES_ENABLE` and I'm still getting network_ups_tools_ups_status metrics. Maybe give that a try.

I am, however, not able to grab battery runtime for whatever reason. Though I may need to tweak my variable names.

DRuggeri, 2025-01-10:
Hey there, @maybesheforgot  - thanks for sharing this. At first blush, alphabetical order or not sounds totally... irrelevant. I'll take a look to see if anything stands out. I'm curious how you landed on this solution? What other permutations did you try that ultimately led here?

DRuggeri, 2025-01-10:
Circling back - I'm able to see the expected status metrics if I use `NUT_EXPORTER_VARIABLES="ups.realpower,ups.load,output.voltage,outlet.power,battery.runtime,battery.charge,ups.status,battery.charge.low,battery.runtime.low"`. At least as far as the original report from @varac, I believe the reason that you didn't see status is because `ups.status` was just not in the list of requested variables. @maybesheforgot, similarly, I can get status information if I put the `ups.status` anywhere in the list - so maybe the alphabetical ordering was just coincidence?

Either way, I just pushed a new commit that will issue a warning on startup if `ups.status` is not in the list of requested metrics since that's one that I anticipate just about everyone would expect to see.

closed 2025-01-10 by DRuggeri

maybesheforgot, 2025-01-10:
Ah, must’ve been coincidental then @DRuggeri. I was looking at the values from another NUT exporter utility I had running on my machine and all the values were listed alphabetically. And since that was what stood out compared to the way they were randomly listed by @varac, I figured that might be why. I want to say I originally only tested a few variables alphabetically at first, but it’s possible ups.status was among those in the test. 

Thanks for following up and further clarifying this!

varac, 2025-01-23:
Sorry for the late feedback. I can now export all important metrics thanks to @maybesheforgot !