DRuggeri/nut_exporter #54: network_ups_tools_battery_charge metric is missing when using NUT_EXPORTER_VARIABLES

a15355447898a, 2025-02-07:
I also have a similar problem to [network_ups_tools_ups_status metric is missing when using NUT_EXPORTER_VARIABLES](https://github.com/DRuggeri/nut_exporter/issues/52), but what I'm missing is battery.charge

I've tried switching the order, but that didn't fix the problem.

My UPS NUT output values are as follows.
```
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
battery.runtime: 1104
battery.type: PbAc
device.mfr: EATON
device.model: SANTAK TG-BOX 850
device.serial: Blank
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2
driver.version.data: MGE HID 1.46
driver.version.internal: 0.53
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: no
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: yes
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 220
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02.08.0010
ups.load: 27
ups.mfr: EATON
ups.model: SANTAK TG-BOX 850
ups.power.nominal: 850
ups.productid: ffff
ups.realpower: 184
ups.serial: Blank
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.type: offline / line interactive
ups.vendorid: 0463  
```

If my Docker Compose is configured like this
```
    nut_exporter:
        container_name: nut_exporter
        image: ghcr.io/druggeri/nut_exporter:latest
        restart: always
        ports:
          - "10443:9199"
        environment:
          - NUT_EXPORTER_SERVER=192.168.1.90
          - NUT_EXPORTER_USERNAME=monuser
          - NUT_EXPORTER_PASSWORD=secret
          - NUT_EXPORTER_VARIABLES="ups.load,ups.status,battery.runtime,output.voltage,ups.realpower,battery.charge"
```

Output as follows:
```
# HELP network_ups_tools_battery_runtime Battery runtime (seconds) (battery.runtime)
# TYPE network_ups_tools_battery_runtime gauge
network_ups_tools_battery_runtime 1104
# HELP network_ups_tools_device_info UPS Device information
# TYPE network_ups_tools_device_info gauge
network_ups_tools_device_info{contact="",description="",location="",macaddr="",mfr="EATON",model="SANTAK TG-BOX 850",part="",serial="Blank",type="ups"} 1
# HELP network_ups_tools_output_voltage Output voltage (V) (output.voltage)
# TYPE network_ups_tools_output_voltage gauge
network_ups_tools_output_voltage 230
# HELP network_ups_tools_ups_realpower Current value of real power (W) (ups.realpower)
# TYPE network_ups_tools_ups_realpower gauge
network_ups_tools_ups_realpower 177
# HELP network_ups_tools_ups_status UPS status (ups.status)
# TYPE network_ups_tools_ups_status gauge
network_ups_tools_ups_status{flag="BOOST"} 0
network_ups_tools_ups_status{flag="BYPASS"} 0
network_ups_tools_ups_status{flag="CAL"} 0
network_ups_tools_ups_status{flag="CHRG"} 0
network_ups_tools_ups_status{flag="DISCHRG"} 0
network_ups_tools_ups_status{flag="FSD"} 0
network_ups_tools_ups_status{flag="HB"} 0
network_ups_tools_ups_status{flag="LB"} 0
network_ups_tools_ups_status{flag="OB"} 0
network_ups_tools_ups_status{flag="OFF"} 0
network_ups_tools_ups_status{flag="OL"} 1
network_ups_tools_ups_status{flag="OVER"} 0
network_ups_tools_ups_status{flag="RB"} 0
network_ups_tools_ups_status{flag="SD"} 0
network_ups_tools_ups_status{flag="TRIM"} 0
# HELP promhttp_metric_handler_errors_total Total number of internal errors encountered by the promhttp metric handler.
# TYPE promhttp_metric_handler_errors_total counter
promhttp_metric_handler_errors_total{cause="encoding"} 0
promhttp_metric_handler_errors_total{cause="gathering"} 0
# HELP promhttp_metric_handler_requests_in_flight Current number of scrapes being served.
# TYPE promhttp_metric_handler_requests_in_flight gauge
promhttp_metric_handler_requests_in_flight 1
# HELP promhttp_metric_handler_requests_total Total number of scrapes by HTTP status code.
# TYPE promhttp_metric_handler_requests_total counter
promhttp_metric_handler_requests_total{code="200"} 15
promhttp_metric_handler_requests_total{code="500"} 0
promhttp_metric_handler_requests_total{code="503"} 0
```

But if I don't set the `NUT_EXPORTER_VARIABLES` variable
```
    nut_exporter:
        container_name: nut_exporter
        image: ghcr.io/druggeri/nut_exporter:latest
        restart: always
        ports:
          - "10443:9199"
        environment:
          - NUT_EXPORTER_SERVER=192.168.1.90
          - NUT_EXPORTER_USERNAME=monuser
          - NUT_EXPORTER_PASSWORD=secret
          #- NUT_EXPORTER_VARIABLES="ups.load,ups.status,battery.runtime,output.voltage,ups.realpower,battery.charge"
```

Output as follows:
```
# HELP network_ups_tools_battery_charge Battery charge (percent of full) (battery.charge)
# TYPE network_ups_tools_battery_charge gauge
network_ups_tools_battery_charge 100
# HELP network_ups_tools_device_info UPS Device information
# TYPE network_ups_tools_device_info gauge
network_ups_tools_device_info{contact="",description="",location="",macaddr="",mfr="EATON",model="SANTAK TG-BOX 850",part="",serial="Blank",type="ups"} 1
# HELP network_ups_tools_ups_load Load on UPS (percent of full) (ups.load)
# TYPE network_ups_tools_ups_load gauge
network_ups_tools_ups_load 26
# HELP network_ups_tools_ups_status UPS status (ups.status)
# TYPE network_ups_tools_ups_status gauge
network_ups_tools_ups_status{flag="BOOST"} 0
network_ups_tools_ups_status{flag="BYPASS"} 0
network_ups_tools_ups_status{flag="CAL"} 0
network_ups_tools_ups_status{flag="CHRG"} 0
network_ups_tools_ups_status{flag="DISCHRG"} 0
network_ups_tools_ups_status{flag="FSD"} 0
network_ups_tools_ups_status{flag="HB"} 0
network_ups_tools_ups_status{flag="LB"} 0
network_ups_tools_ups_status{flag="OB"} 0
network_ups_tools_ups_status{flag="OFF"} 0
network_ups_tools_ups_status{flag="OL"} 1
network_ups_tools_ups_status{flag="OVER"} 0
network_ups_tools_ups_status{flag="RB"} 0
network_ups_tools_ups_status{flag="SD"} 0
network_ups_tools_ups_status{flag="TRIM"} 0
# HELP promhttp_metric_handler_errors_total Total number of internal errors encountered by the promhttp metric handler.
# TYPE promhttp_metric_handler_errors_total counter
promhttp_metric_handler_errors_total{cause="encoding"} 0
promhttp_metric_handler_errors_total{cause="gathering"} 0
# HELP promhttp_metric_handler_requests_in_flight Current number of scrapes being served.
# TYPE promhttp_metric_handler_requests_in_flight gauge
promhttp_metric_handler_requests_in_flight 1
# HELP promhttp_metric_handler_requests_total Total number of scrapes by HTTP status code.
# TYPE promhttp_metric_handler_requests_total counter
promhttp_metric_handler_requests_total{code="200"} 1
promhttp_metric_handler_requests_total{code="500"} 0
promhttp_metric_handler_requests_total{code="503"} 0
```

a15355447898a, 2025-02-09:
This environment variable fixed my problem, but I still don't know why.
`input.transfer.low,output.voltage,battery.charge,ups.realpower,ups.load,ups.status,battery.runtime,input.transfer.high`

a15355447898a, 2025-02-13:
> Interesting. Because the default list of variables scraped includes battery.charge, I think disabling override of the ENV var is why you see it in the second case. However, I can't think of any reason why the first string you used (`ups.load,ups.status,battery.runtime,output.voltage,ups.realpower,battery.charge`) didn't show battery.charge, but the second string you used (`input.transfer.low,output.voltage,battery.charge,ups.realpower,ups.load,ups.status,battery.runtime,input.transfer.high`) did. Both include that variable.
> 
> I know you got it working, but if possible, sharing debug output with the first string will help demystify what may have happened. The logic to parse the list of variables is pretty simple (i.e... if one works, all should work) so I wonder if there may have been something else within the exporter that led to that missing variable.

I used this Docker Compose file
```yml
services:
  nut_exporter:
    container_name: nut_exporter
    image: ghcr.io/druggeri/nut_exporter:latest
    restart: always
    ports:
      - "10443:9199"
    environment:
      - NUT_EXPORTER_SERVER=192.168.1.90
      - NUT_EXPORTER_USERNAME=monuser
      - NUT_EXPORTER_PASSWORD=secret
      - NUT_EXPORTER_VARIABLES="ups.load,ups.status,battery.runtime,output.voltage,ups.realpower,battery.charge"
    command: ["--log.level=debug"]
    networks:
      - homelab-network

networks:
  homelab-network:
    external: true
```

Output on the webpage as follows
```
# HELP network_ups_tools_battery_runtime Battery runtime (seconds) (battery.runtime)
# TYPE network_ups_tools_battery_runtime gauge
network_ups_tools_battery_runtime 1152
# HELP network_ups_tools_device_info UPS Device information
# TYPE network_ups_tools_device_info gauge
network_ups_tools_device_info{contact="",description="",location="",macaddr="",mfr="EATON",model="SANTAK TG-BOX 850",part="",serial="Blank",type="ups"} 1
# HELP network_ups_tools_output_voltage Output voltage (V) (output.voltage)
# TYPE network_ups_tools_output_voltage gauge
network_ups_tools_output_voltage 230
# HELP network_ups_tools_ups_realpower Current value of real power (W) (ups.realpower)
# TYPE network_ups_tools_ups_realpower gauge
network_ups_tools_ups_realpower 184
# HELP network_ups_tools_ups_status UPS status (ups.status)
# TYPE network_ups_tools_ups_status gauge
network_ups_tools_ups_status{flag="BOOST"} 0
network_ups_tools_ups_status{flag="BYPASS"} 0
network_ups_tools_ups_status{flag="CAL"} 0
network_ups_tools_ups_status{flag="CHRG"} 0
network_ups_tools_ups_status{flag="DISCHRG"} 0
network_ups_tools_ups_status{flag="FSD"} 0
network_ups_tools_ups_status{flag="HB"} 0
network_ups_tools_ups_status{flag="LB"} 0
network_ups_tools_ups_status{flag="OB"} 0
network_ups_tools_ups_status{flag="OFF"} 0
network_ups_tools_ups_status{flag="OL"} 1
network_ups_tools_ups_status{flag="OVER"} 0
network_ups_tools_ups_status{flag="RB"} 0
network_ups_tools_ups_status{flag="SD"} 0
network_ups_tools_ups_status{flag="TRIM"} 0
# HELP promhttp_metric_handler_errors_total Total number of internal errors encountered by the promhttp metric handler.
# TYPE promhttp_metric_handler_errors_total counter
promhttp_metric_handler_errors_total{cause="encoding"} 0
promhttp_metric_handler_errors_total{cause="gathering"} 0
# HELP promhttp_metric_handler_requests_in_flight Current number of scrapes being served.
# TYPE promhttp_metric_handler_requests_in_flight gauge
promhttp_metric_handler_requests_in_flight 1
# HELP promhttp_metric_handler_requests_total Total number of scrapes by HTTP status code.
# TYPE promhttp_metric_handler_requests_total counter
promhttp_metric_handler_requests_total{code="200"} 0
promhttp_metric_handler_requests_total{code="500"} 0
promhttp_metric_handler_requests_total{code="503"} 0
[... 80 lines cut ...]
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:176 level=debug variable_name=ups.load value=27 type=INTEGER description="Load on UPS (percent of full)" writeable=false maximum_length=0 original_type=NUMBER
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:269 level=debug msg="Export the variable? false" count=6 variables="unsupported value type"
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:176 level=debug variable_name=ups.mfr value=EATON type=STRING description="UPS manufacturer" writeable=false maximum_length=0 original_type=NUMBER
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:269 level=debug msg="Export the variable? false" count=6 variables="unsupported value type"
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:176 level=debug variable_name=ups.model value="SANTAK TG-BOX 850" type=STRING description="UPS model" writeable=false maximum_length=0 original_type=NUMBER
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:269 level=debug msg="Export the variable? false" count=6 variables="unsupported value type"
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:176 level=debug variable_name=ups.power.nominal value=850 type=INTEGER description="UPS power rating (VA)" writeable=false maximum_length=0 original_type=NUMBER
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:269 level=debug msg="Export the variable? false" count=6 variables="unsupported value type"
ts=2025-02-13T12:33:59.240Z caller=nut_collector.go:176 level=debug variable_name=ups.productid value=ffff type=STRING description="Product ID for USB devices" writeable=false maximum_length=0 original_type=NUMBER
[... cut ...]
ts=2025-02-13T12:33:59.242Z caller=nut_collector.go:269 level=debug msg="Export the variable? false" count=6 variables="unsupported value type"
ts=2025-02-13T12:33:59.242Z caller=nut_collector.go:176 level=debug variable_name=ups.vendorid value=463 type=INTEGER description="Vendor ID for USB devices" writeable=false maximum_length=0 original_type=NUMBER
ts=2025-02-13T12:33:59.242Z caller=nut_collector.go:269 level=debug msg="Export the variable? false" count=6 variables="unsupported value type"
```

DRuggeri, 2025-03-26:
Interestingly, a similar problem (not totally diagnosed either) is reported in #50. I just created a new release of the collector that switches to using slog and also to log the exact variables provided. If you're so inclined, it'd be great if you could test it out!

a15355447898a, 2025-03-31:
> Interestingly, a similar problem (not totally diagnosed either) is reported in [#50](https://github.com/DRuggeri/nut_exporter/issues/50). I just created a new release of the collector that switches to using slog and also to log the exact variables provided. My suspicion is that the collector is getting the string "unsupported value type" as input someway somehow. If you're so inclined, it'd be great if you could test it out!

The new version of the docker container works fine for me, no more problems!
Using this docker compose
```yaml
services:
  nut_exporter:
    container_name: nut_exporter
    image: ghcr.io/druggeri/nut_exporter:latest
    restart: always
    ports:
      - "10440:9199"
    environment:
      - NUT_EXPORTER_SERVER=192.168.1.90
      - NUT_EXPORTER_USERNAME=monuser
      - NUT_EXPORTER_PASSWORD=secret
      - NUT_EXPORTER_VARIABLES=ups.load,ups.status,battery.runtime,output.voltage,ups.realpower,battery.charge
    networks:
      - homelab-network
  
networks:
  homelab-network:
    external: true
```
The following data was obtained
```
# HELP network_ups_tools_battery_charge Battery charge (percent of full) (battery.charge)
# TYPE network_ups_tools_battery_charge gauge
network_ups_tools_battery_charge 100
# HELP network_ups_tools_battery_runtime Battery runtime (seconds) (battery.runtime)
# TYPE network_ups_tools_battery_runtime gauge
network_ups_tools_battery_runtime 1152
# HELP network_ups_tools_device_info UPS Device information
# TYPE network_ups_tools_device_info gauge
network_ups_tools_device_info{contact="",description="",location="",macaddr="",mfr="EATON",model="SANTAK TG-BOX 850",part="",serial="Blank",type="ups"} 1
# HELP network_ups_tools_output_voltage Output voltage (V) (output.voltage)
# TYPE network_ups_tools_output_voltage gauge
network_ups_tools_output_voltage 230
# HELP network_ups_tools_ups_load Load on UPS (percent of full) (ups.load)
# TYPE network_ups_tools_ups_load gauge
network_ups_tools_ups_load 23
# HELP network_ups_tools_ups_realpower Current value of real power (W) (ups.realpower)
# TYPE network_ups_tools_ups_realpower gauge
network_ups_tools_ups_realpower 156
# HELP network_ups_tools_ups_status UPS status (ups.status)
# TYPE network_ups_tools_ups_status gauge
network_ups_tools_ups_status{flag="BOOST"} 0
network_ups_tools_ups_status{flag="BYPASS"} 0
network_ups_tools_ups_status{flag="CAL"} 0
network_ups_tools_ups_status{flag="CHRG"} 0
network_ups_tools_ups_status{flag="DISCHRG"} 0
network_ups_tools_ups_status{flag="FSD"} 0
network_ups_tools_ups_status{flag="HB"} 0
network_ups_tools_ups_status{flag="LB"} 0
network_ups_tools_ups_status{flag="OB"} 0
network_ups_tools_ups_status{flag="OFF"} 0
network_ups_tools_ups_status{flag="OL"} 1
network_ups_tools_ups_status{flag="OVER"} 0
network_ups_tools_ups_status{flag="RB"} 0
network_ups_tools_ups_status{flag="SD"} 0
network_ups_tools_ups_status{flag="TRIM"} 0
# HELP promhttp_metric_handler_errors_total Total number of internal errors encountered by the promhttp metric handler.
# TYPE promhttp_metric_handler_errors_total counter
promhttp_metric_handler_errors_total{cause="encoding"} 0
promhttp_metric_handler_errors_total{cause="gathering"} 0
# HELP promhttp_metric_handler_requests_in_flight Current number of scrapes being served.
# TYPE promhttp_metric_handler_requests_in_flight gauge
promhttp_metric_handler_requests_in_flight 1
# HELP promhttp_metric_handler_requests_total Total number of scrapes by HTTP status code.
# TYPE promhttp_metric_handler_requests_total counter
promhttp_metric_handler_requests_total{code="200"} 0
promhttp_metric_handler_requests_total{code="500"} 0
promhttp_metric_handler_requests_total{code="503"} 0
```

vtmocanu, 2025-06-08:
I have a similar problem, latest tag does not solve it for me:
```
    - name: NUT_EXPORTER_VARIABLES
      value: "battery.runtime,battery.charge,battery.voltage,battery.voltage.nominal,input.voltage,input.voltage.nominal,ups.load,ups.status"
```
```
time=2025-06-08T10:55:45.453+03:00 level=DEBUG msg=variable_name battery.runtime=value !BADKEY=2100 type=INTEGER description="Battery runtime (seconds)" writeable=false maximum_length=0 original_type=NUMBER
time=2025-06-08T10:55:45.453+03:00 level=DEBUG msg="Export the variable? true"
time=2025-06-08T10:55:45.453+03:00 level=DEBUG msg=variable_name battery.runtime.low=value !BADKEY=120 type=INTEGER description="Remaining battery runtime when UPS switches to LB (seconds)" writeable=true maximum_length=10 original_type=STRING
time=2025-06-08T10:55:45.453+03:00 level=DEBUG msg="Export the variable? false" count=8 variables=battery.runtime,battery.charge,battery.voltage,battery.voltage.nominal,input.voltage,input.voltage.nominal,ups.load,ups.status
```

`network_ups_tools_battery_runtime` is missing from the output, I don't understand why, because I see export is true.
anything I can do to debug further?
thank you!