DRuggeri/nut_exporter #55: Healthcheck

vbaidak, 2025-02-14:
Hi!

Could you please provide an example for the `healthcheck` for Docker?

DRuggeri, 2025-07-03:
Hey there, @vbaidak  - sorry for the delay in responding. I somehow missed this one.

It should be safe to hit the /metrics endpoint as a health observer since these metrics are just the built-in Go prometheus SDK metrics (all coming from memory which should be inexpensive to scrape).

If that works for your use case, would you consider a PR to add to do the docs?

vbaidak, 2025-07-03:
Hi @DRuggeri ,

I could be mistaken, but as far as I know, the golang:1-alpine image is quite minimal and doesn't include any utilities that would allow querying an endpoint in a HEALTHCHECK directive

DRuggeri, 2025-07-03:
Ah, apologies - I was mixing up Kubernetes health check (HTTP probe) with Docker's health check (command execution). I see that the alpine image used by this exporter includes `wget`.

This ought to work in that case:
```
wget -O /dev/null https://127.0.0.1:3493/metrics
```

vbaidak, 2025-07-03:
I don't think it is present. Here is my command:

```bash
$ docker exec nut-exporter wget
OCI runtime exec failed: exec failed: unable to start container process: exec: "wget": executable file not found in $PATH: unknown
```

and I'm using it in compose also like this:
```yaml
    healthcheck:
      test: ["CMD", "wget", "-q", "--spider", "http://localhost:3493/metrics"]
      interval: 30s
      timeout: 5s
      retries: 3
```

DRuggeri, 2026-03-03:
I just confirmed the updated GH action did it - the container image now contains the base alpine packages, making `wget` a viable option for healthcheck!

mehalter, 2026-03-03:
Could we add in an official built in healthcheck? I had the line in my PR that I had opened