DRuggeri/nut_exporter #58: helm chart not published

vtmocanu, 2025-06-08:
It looks like the Chart is not published:

```
helm repo add nut-exporter https://github.com/DRuggeri/nut_exporter

Error: looks like "https://github.com/DRuggeri/nut_exporter" is not a valid chart repository or cannot be reached: failed to fetch https://github.com/DRuggeri/nut_exporter/index.yaml : 404 Not Found
```

Thank you for sharing this exporter!

peterrosell, 2025-06-09:
I also tried to use the helm chart, but got the same error. I see that the `release-helm` workflow triggers on push on the `main`, but this repo uses `master` branch. Just changing it to `master` will trigger the workflow and hopefully create what's needed.

DRuggeri, 2025-07-03:
Awesome, @peterrosell - nice and easy fix. Thanks for the PR!

Sorry for the delay merging - I've been on summer travel holiday :-)

closed 2025-07-03 by DRuggeri

vtmocanu, 2025-07-03:
Looks like it's still not working?

vtmocanu, 2025-08-27:
@DRuggeri can you pls reopen this one?

guillomep, 2025-10-29:
@DRuggeri I think you forgot to apply the modification

DRuggeri, 2025-11-07:
Issue has been reopened, but I see the requested change is in place 🤔

It looks like it [ran on the release from two weeks ago](https://github.com/DRuggeri/nut_exporter/actions/runs/18766838901/job/53543404041), but it reportedly found no changes in the tag?

```
Run helm/chart-releaser-action@v1.6.0
Run owner=$(cut -d '/' -f 1 <<< "$GITHUB_REPOSITORY")
Looking up latest tag...
Discovering changed charts since 'v3.2.1'...
Nothing to do. No chart changes detected.
Run helm/chart-releaser-action@v1.6.0
Run owner=$(cut -d '/' -f 1 <<< "$GITHUB_REPOSITORY")
Looking up latest tag...
Discovering changed charts since 'v3.2.1'...
Nothing to do. No chart changes detected.
```

Apologies, but I'm not very familiar with this workflow action and what it expects. Is it unhappy that the chart itself hasn't changed since it was once released, or... ?
Pointers definitely appreciated.

guillomep, 2025-11-07:
Weird, it seems that your workflow is right https://helm.sh/docs/howto/chart_releaser_action/

You do not have a gh-pages branch, but I don't think this is a problem. Also it seems that the step "Install helm" is unnecessary.